headtracker/HeadTracker #158: Futaba 8FG - PPM - can't use channel after 8

jerem94000, 2023-09-12:
Hello

For the Arduino HeadTracker, i would need to use Channel 10, 11 and 12.
To do this, I need to activate "12CH" for the trainer in the 8FG radio.
But as soon as I  switch "8CH" to "12CH" on the Trainer, I can note receive correctly the signal from the Arduino, which directly appears to its max.

Do you know why ?

Thanks in advance :-).

[image]
[image]

rotorman, 2023-09-12:
See: https://www.rcgroups.com/forums/showpost.php?p=31693706&postcount=6285

jerem94000, 2023-09-12:
Aahh ... thank you for this link.
I didn't succeed to find any information on this subject.

Indeed, To be able use more than 8 channels in the 8FG : 
- I set in the Radio : MLT2 for the frequency
- and in the Trainer configuration : 12CH

And when I activate the trainer swith on the radio, it capts a signal ... but the 3 channels (10, 11 and 12) direclty go to the max and totally not react to the move of the Arduino.

rotorman, 2023-09-12:
Get a hold to a scope or a logic analyzer. Then put your Futaba into 12 channel trainer mode as student radio and capture and post the signal. We can extract from that the timing difference to "normal" 8 channel PPM pulses.

jerem94000, 2023-09-12:
> Get a hold to a scope or a logic analyzer. 

Could you please give me a link which shows me what is this scope ?
Thank you :-)

rotorman, 2023-09-12:
Something like this (or similar): https://www.tek.com/en/products/oscilloscopes

jerem94000, 2023-09-12:
Ah yes of course.
I already used it ... 30 yeas ago, when I was in high school !!
A bit expensive  for me this test...

rotorman, 2023-09-12:
You do not need to buy it. Just find someone who has oscilloscope, or even better a logic analyzer, like e.g. Saleae, and then ask to use it for 5 minutes.

dlktdr, 2024-02-24:
> Aahh ... thank you for this link. I didn't succeed to find any information on this subject.
> 
> Indeed, To be able use more than 8 channels in the 8FG :
> 
> * I set in the Radio : MLT2 for the frequency
> * and in the Trainer configuration : 12CH
> 
> But when I activate the trainer switch on the radio, it capts a signal ... but the 3 channels (10, 11 and 12) direclty go to the max and totally not react to the move of the Arduino (as we can see on the first copy screen upper.
> 
> Strange not find other posts talking about this subjet. I may not be the first to want to use channel above Channel 8 ... moreover if we want to use Pan + Tilt + Roll.

Did you also increase the channel count of the PPM output in the GUI?  Since there are 12 channels you may need to also increase the frame lenght.
[image]

closed 2024-07-30 by dlktdr