headtracker/HeadTracker #161: ABX00069Drift problems

tianbin4729, 2024-01-11:
When using ABX00069, I found that rapid movement can cause yaw axis drift, which often occurs. Moreover, ABX00069 cannot perform mag calibration。

dlktdr, 2024-02-24:
I have some work here I think there is a change affecting newer versions for some.

Not sure about the mag calibration. If you check the box for use Magnetometer in the box here 
[image]
close the calibration window and check under diagnostic display, do you see the sensor responding
[image]

tianbin4729, 2024-02-24:
[image]
[image]
The data of magx magy has not changed

closed 2024-07-30 by dlktdr