headtracker/HeadTracker #163: Windows 11 support

bxrogers, 2024-02-24:
The graph at the bottom is not viewable.  I have no idea whether the sensor are working .  I did a calibration, but couldn’t tell if it helped or hurt.

all the fonts seem huge.  It reminds me of trying to open windows 95 apps in windows 10.  

is this normal?  I’m not sure there is much to do except find another use for all the hardware :)

I’m on the latest version…

dlktdr, 2024-02-24:
[image]

How it looks for me on Win 11. Nothing too out of place. Your using v2.2 and not a very old one?

Can you post a screenshot too.

dlktdr, 2024-05-16:
Closing for lack of feedback

closed 2024-05-16 by dlktdr