headtracker/HeadTracker #166: No IMU output on nano 33 BLE Rev2 with fw 2.2

totorus, 2024-04-11:
Hello

I have just installed V2.2 on a NANO33 BLE Rev2 out of the box but I have no movement info.
I can see the green light on "Gyrometer Calibrated", the yellow diamonds in the pan/tilt/roll range but no activity when I move the card.
I tried reinstalling the firmware but it didn't change anything.

Has anyone got any ideas?

totorus, 2024-04-11:
Great, everything's working now!
Thanks for the quick response :)

dlktdr, 2024-04-12:
Just found a pretty major flaw in that one if you try to use board rotation. Might want to try this one.

https://github.com/dlktdr/HeadTracker/actions/runs/8656253193

totorus, 2024-04-12:
with this release I can't save my settings.
I've reset the card after updating it with the latest firmware and it's impossible to save. The save button works but after rebooting the settings disappear.
With the first firmware it keeps my settings.
Is it the same for you?

dlktdr, 2024-04-12:
Probably something i missed. Will check tonight.

dlktdr, 2024-04-13:
Yeah put the update call in the wrong spot. 

https://github.com/dlktdr/HeadTracker/commit/210c61d767caa954eb7a80c4477897bb5f8748d4

dlktdr, 2024-05-16:
Solved in issue #169.

closed 2024-05-16 by dlktdr

cayote64, 2024-07-09:
I have downloaded the REV 22 firmware from above, but the 2.4 GUI artefact has expired so still cant get it to work.