headtracker/HeadTracker #169: arduino nano 33 ble FW 2_4pre no BLE connection  possible, but gets deleted with reboot

boomerschnecke, 2024-04-18:
the BLE settings could not  be stored specially for BLE, reset disables BLE

dlktdr, 2024-04-19:
Did you try the most recent action build. There was an issue saving settings. I think this is solved here, https://github.com/dlktdr/HeadTracker/commit/210c61d767caa954eb7a80c4477897bb5f8748d4

boomerschnecke, 2024-04-19:
So I used the latest available FW from your link at Discord. I tried this setup for HeadTracker with XIOA at Radio Side and with arduino nano 33 ble at Radio Side. Both have a Bluetooth conncection to Radio via UART. Both show this reset

For this FW for arduino nano ble 2_4pre. I could set the Bluetooth settings. but after storing the settings and doing a reboot its gone

dlktdr, 2024-04-19:
I made a fix after that, found the saving problem.

Action for FW is here https://github.com/dlktdr/HeadTracker/actions/runs/8671926220
Action for GUI is here https://github.com/dlktdr/HeadTracker/actions/runs/8671926216

dschiedsch, 2024-05-01:
Hi
The saving problem seems fixed but the bluetooth connection is not re-established after a reboot
Bt Mode is correct after a restart but status is Disabled and Address also shows BT_DISABLED

dschiedsch, 2024-05-01:
Did some more digging and it looks like the bluetooth mode is stored correctly and btmode in the settings display also shows e.g. 1 but bluetooth seems not to be initialized if that is the case.

Dumbtoom, 2024-05-12:
Hello, where can I download the FW for arduino nano ble 2_4pre? I only find verion 2.2
Thanks and best regards,
Tom

boomerschnecke, 2024-05-12:
> connection possible, but gets deleted with reboot (Issue #169)

Chapter test-firmware

dlktdr, 2024-05-12:
These are the most recent builds,

https://github.com/dlktdr/HeadTracker/actions/workflows/build-firmware.yml

Ones at the top are the most recent. You need to be logged into Github to download them

dschiedsch, 2024-05-12:
Did a bit more testing this weekend and after a lot of back and forth with different settings i managed to get it to save and work, once.
I tried recreating this on a 2nd board with exact same settings but without success.
This sounds to me like a timing issue.
Is it possible that you @dlktdr tested with a debug compiled version that has slightly different timings?

I will try to compile the gui as per instructions and see if a debug compile works differently

dlktdr, 2024-05-12:
Spent some more time on this and found that save to flash wasn't working on every write. Something to do with the new version of Zephyr. Reworked it from the samples and it seems to save properly. 

The action is compiling right now. Please give it a test when it's done. Be sure to use the 246068c GUI as well, there was also an error in that.

dschiedsch, 2024-05-13:
Did try on 3 boards shortly and it looks like it is working properly now.
Boards come up with and without pc connection and bluetooth connection is re-established without issue.

dschiedsch, 2024-05-13:
Tried on 3 more boards and about 20 reboot cycles, I don't have any issues anymore.
Seems to me that this is fixed.

Dumbtoom, 2024-05-15:
Hello,
I was lucky to get hands on an old ble 33 board (rev1). Everything is working just fine with this board. Also I got a sense version (rev2) and it apears to work. Only the rev 2 no sensor version is not working, no matter which latest update of firmware and gui I am using. For me, using the old board - problem solved :-)

dlktdr, 2024-05-16:
> Hello, I was lucky to get hands on an old ble 33 board (rev1). Everything is working just fine with this board. Also I got a sense version (rev2) and it apears to work. Only the rev 2 no sensor [...]

Can you please confirm what builds you have tried. Only the most recent seem to be working properly on Rev2. You can download the most recent from Actions, links above https://github.com/dlktdr/HeadTracker/issues/169#issuecomment-2106297284. H  https://headtracker.gitbook.io/head-tracker-v2.2/support/development/git-workflow-and-actions

The most recent release 2.2 does not work properly on the Rev 2 boards.

Once running you should see the two SHA#'s here. 
[image]

dschiedsch, 2024-05-16:
My testing above was done on nano 33 ble non sense boards (ABX00071)
running 43c0c37 firmware and for me it was working fine on a total of 6 boards
If you're using that fw version which functionality is not working for you?

boomerschnecke, 2024-05-16:
Hello all,

could you please tell we, where I could directly download all this different versions?

Best regards

dlktdr, 2024-05-16:
@boomerschnecke 

You can download them from the Actions page of Github. The downloads are called Artifacts they are at the bottom of every build. You must be logged into Github to download them.

Start reading at Workflows,
https://headtracker.gitbook.io/head-tracker-v2.2/support/development/git-workflow-and-actions

I'm going to close this issue as the problem at hand appears to be solved.

closed 2024-05-16 by dlktdr