headtracker/HeadTracker #172: Drift on pan, wired to trainer port, Help!

LeonRemidy, 2024-06-14:
Hi folks 

I'm a bit of a rookie with this kind of thing but I got the board soldered and installed via a aux cable to my old non-bluetooth x9d.. works almost as it should but I keep having pan drift issues.. 

Can anyone please advise? 

Many thanks

Sibi4711, 2024-06-19:
if you have drift then either re-calibrate or try not to  look down while panning. Try to keep your head straight so the HT is oriented correctly.

dlktdr, 2024-06-19:
Can you please specify what version of code you are running. There have been some issues with 2.2. I would try going back to version 2.1 and see if the issue persists. 

Can also try the newest 2.4 but you will have to try the most recent builds from Github Actions

closed 2025-03-15 by dlktdr