headtracker/HeadTracker #173: hello impossible to calibrat eand get IMU signal

samsab1, 2024-07-27:
hello
Hello everyone

I just flashed my arduino cc nano ble 33rev2 ABX 00071 with headtracker release 2.2. the flashing seems perfectly normal
the calibration phase seems to be going correctly, but I ultimately believe that it is never saved correctly although I ask for this calibration to be saved.

finally, the curves remain perfectly flat and smooth when I move the ARDUINO board. who could give me light?

bluetooth seems to be ok when connecting to https://headtracker.net/v22/
[image]
[image]
[image]
[image]
[image]
[image]
[image]

samsab1, 2024-07-27:
[image]
[image]

dlktdr, 2024-07-27:
Rev2 needs to use the new firmware and gui.

https://github.com/dlktdr/HeadTracker/actions/runs/10010869506
https://github.com/dlktdr/HeadTracker/actions/runs/9124117598

You can find the most recent ones under actions. To upload the firmware, you have to select local file and use the extracted one from the firmware build

samsab1, 2024-07-27:
many thanks for your answer.
so my ABX00071 remains usefull? but with an other firmware; right?

can you be more accurate in your answer, i m a newbie with arduino, and a french guy who is very not fluent in english.
do you mean that i keep the frontoffice software "headtracker 2.2" but i use the firmware 
[image]

why did you give 2 links? how can i understand those 2 links?
best regards

samsab1, 2024-07-28:
ok Cliff, it was so easy, thanks a lot , it seems to be ok
very best regards

Aaronsss, 2024-07-29:
I have just ran into the same issue but dont seem to be able to resolve it with the above solution 
[image]

dlktdr, 2024-07-30:
> I have just ran into the same issue but dont seem to be able to resolve it with the above solution ![image](https://private-user-images.githubusercontent.com/23297034/353249757-db245e24-5cb0-4d94-9443-5b24916894dd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjIzNTUyOTEsIm5iZiI6MTcyMjM1NDk5MSwicGF0aCI6Ii8yMzI5NzAzNC8zNTMyNDk3NTctZGIyNDVlMjQtNWNiMC00ZDk0LTk0NDMtNWIyNDkxNjg5NGRkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzMwVDE1NTYzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZiZjE3MjlmZWFkMzI4ZTVhYzQxMDYxOTM0Y2E2OTU2NGU1OGRkN2QzNjViMzA0MWNiMGQyNmY1OTQxODZiNjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.5WoKfVNbw6v2EHQ5IPGzlYNytYZTqlsmEjTB_3UadYA)
> 
> It is a brand new unsoldered board but I presume that doesn't matter?
> 
> It looks like it's getting no data from the Arduino for the gyro or accelerometer during cal none of the values move for either accelerometer or magnetometer

When you flashed it you picked the firmware labelled rev2?

You have the same abx00071 as well?

Aaronsss, 2024-07-30:
Yes to both

dlktdr, 2024-07-30:
Could you program a basic Arduino test program to it to see if the IMU is actually working on your board or if it's an issue on my end.

It might also help if you could put the a copy of what the serial monitor shows after a reboot of the board. Then put it in a text file and upload it here.

Aaronsss, 2024-07-30:
I tried this script and get the following message over serial port https://docs.arduino.cc/tutorials/nano-33-ble-rev2/imu-gyroscope/

Started
Failed to initialize IMU!

Output from the serial port with the headtracker code 
[serial boot.txt](https://github.com/user-attachments/files/16433836/serial.boot.txt)

Broken board?

Aaronsss, 2024-07-30:
A bit of further testing if I use Arduino_BMI270_BMM150 1.1.0 it gets a bit further but still does not enter if (IMU.gyroscopeAvailable()) {

Im guessing better checks were introduced from 1.1.1 onwards. It doesnt seem to want to come to life

dlktdr, 2024-07-30:
Hmm yeah didn't detect it in this software either 
```[00065693] <err> htmain: Sense_Init failed``` according to that log.

Kind of sounds like the baord. Can you post a close up photo of the board.

Aaronsss, 2024-07-30:
[image]
[image]

Which ic is the imu? There is a part near the xyz logo that looks chipped. At first I thought it was an indicator but looks like a bga

dlktdr, 2024-07-30:
The one near the xyz is the magnetometer, BMM150. and yeah it sure does look chipped. If it's broken and holding the I2C lines in a low state would cause the IMU(BMI270) to not work too
[image]

Aaronsss, 2024-07-30:
I will try getting the picoscope on the i2c tommorow to see if it's doing anything. If not I will initiate the return to farnell as they actually sent the rev 2 despite advertising the rev 1 part no.

dlktdr, 2024-07-30:
Best of luck.

FYI the I2C is easiest to probe here.
[image]

Aaronsss, 2024-07-31:
straight after reset

When probing the bottom resistor
[image]
[image]
[image]
When probing the top resistor
[image]

Whilst it isnt stuck low I think it is safe to say it is not happy from those waveforms

In any case Farnell already refunded me as they sent the wrong part no. so will try and source another one and then maybe this one can do bluetooth for me :)

Aaronsss, 2024-08-01:
Good news I got a new one and it all seems to be working as expected :)

Damastel, 2024-08-31:
> Rev2 needs to use the development firmware and gui.
> 
> https://github.com/dlktdr/HeadTracker/actions/runs/10010869506 https://github.com/dlktdr/HeadTracker/actions/runs/9124117598
> 
> Bottom of each page is the artifacts(binaries)
> 
> You can find the most recent ones under actions. To upload the firmware, you have to select local file and use the extracted one from the firmware build

Hello, I seem to be having the same issue with my IMU, I think the above suggestion will would but the second link has an "expired" file. Would there be any chance I could get that binary file? Thanks!

dlktdr, 2024-09-03:
> Hello, I seem to be having the same issue with my IMU, I think the above suggestion will work, but the second link has an "expired" file. Would there be any chance I could get that binary file? Thanks!

Since it's the development version and constantly changing you can always find the most recent builds from the actions page. An action automatically compiles binaries on every change.

https://github.com/dlktdr/HeadTracker/actions

On the left pane you can select the action type you want. You will need the most recent firmware and the most recent windows gui. Always check that it is built from the "Master" branch.

[image]

closed 2025-03-15 by dlktdr