headtracker/HeadTracker #174: export roll, tilt, pan values through pins

TalalSultan, 2024-07-29:
hello, i have 2 arduinos ,1 arduino nano BLE which has the software installed in it, and another arduino uno.

all i want is to export the values of the roll, tilt, pan from the nano BLE and read them in the arduino uno, how can i do that ?

rotorman, 2024-07-29:
One option would be to output inverted S.BUS (=std UART) from your nano BLE with HeadTracker firmware and read in the S.BUS signal using your Arduino Uno serial port.

TalalSultan, 2024-07-29:
Thank you for replaying, i did the connection but im getting weird values
code example from sbus i used :
```
/*
* Brian R Taylor
* brian.taylor@bolderflight.com
* 
* Copyright (c) 2022 Bolder Flight Systems Inc
*
* Permission is hereby granted, free of charge, to any person obtaining a copy
* of this software and associated documentation files (the “Software”), to
* deal in the Software without restriction, including without limitation the
* rights to use, copy, modify, merge, publish, distribute, sublicense, and/or
* sell copies of the Software, and to permit persons to whom the Software is
* furnished to do so, subject to the following conditions:
*
* The above copyright notice and this permission notice shall be included in
* all copies or substantial portions of the Software.
*
* THE SOFTWARE IS PROVIDED “AS IS”, WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
* IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
* FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
* AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
* LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING
* FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS
* IN THE SOFTWARE.
*/

#include "sbus.h"

/* SBUS object, reading SBUS */
bfs::SbusRx sbus_rx(&Serial);
/* SBUS object, writing SBUS */
bfs::SbusTx sbus_tx(&Serial);
/* SBUS data */
bfs::SbusData data;

void setup() {
  /* Serial to display data */
  Serial.begin(9600);
  while (!Serial) {}
  /* Begin the SBUS communication */
  sbus_rx.Begin();
  sbus_tx.Begin();
}

void loop () {
  if (sbus_rx.Read()) {
    /* Grab the received data */
    data = sbus_rx.data();
    /* Display the received data */
    for (int8_t i = 0; i < data.NUM_CH; i++) {
      Serial.print(data.ch[i]);
      Serial.print("\t");
    }
    /* Display lost frames and failsafe data */
    Serial.print(data.lost_frame);
    Serial.print("\t");
    Serial.println(data.failsafe);
    /* Set the SBUS TX data to the received data */
    sbus_tx.data(data);
    /* Write the data to the servos */
    sbus_tx.Write();
  }
}

```
i get random outputs
`35iZ��M!35iZ����5��\�q�_��3MУZ=�`

dlktdr, 2024-07-29:
It looks like your using the same Serial device for the communication for sbus and the connection to your computer.

Should use &Serial2 for the SBUS connection. As soon as you call the sbus_rx.Begin() it would change your baud values on Serial(the pc connection)

If the board your trying to get this to with on doesnt

rotorman, 2024-07-30:
I wonder if it might actually work out with only a single serial, as the inverted S.BUS input on Arduino Uno is only using the RX pin and the output stream from the code above the TX line. Try using on the computer terminal the S.BUS serial parameters, thus instead of 9600 baud 8N1, try 100.000 baud and 8E2.

TalalSultan, 2024-07-30:
it worked! i had to use platformIO to open a custom serial port with 100000 baud and 8E2 and i managed to get all the channel readings thankfully, thank you all for helping

closed 2024-07-30 by rotorman