headtracker/HeadTracker #178: Arduino Nano 33 BLE Rev2

deniz-derin, 2024-09-01:
Hello;

I ordered an Arduino Nano 33 BLE for my project, but the seller sent me an Arduino Nano 33 BLE Rev2 board. I uploaded the V2.2 software to the board, but I can't get any axis changes. I'm pretty new to this. Does the software not support the Arduino Nano 33 BLE Rev2 board? If not, is there a chance to add it to the software? I didn't see a title on this subject. I kindly ask you to direct me. Best regards from Turkey and thanks in advance.
[image]

dlktdr, 2024-09-03:
You will need to use the development version to get it working with the Rev2 boards.

Please check the actions tab above, download the most recent firmware build and the most recent Windows gui build. You can find the files to download in the Artifacts at the bottom of the page. 

When you flash, choose local file from the GUI and navigate to the extracted firmwares that were downloaded above. 

Current most recent are,
https://github.com/dlktdr/HeadTracker/actions/runs/10471219932

deniz-derin, 2024-09-12:
Hello dear Dlktdr;

Thank you very much for your support. I will try it quickly.

DavidGom1, 2024-09-12:
Hello, I have an Arduino 33 BLE with firmware version 2.40pre loaded. I find that roll and pan, to one side the number is positive and to the other negative, but tilt to one side is positive but when it has to go down it does so in a regressive manner from 360.

Has this happened to you?

deniz-derin, 2024-09-13:
Hello dear Dlktdr;

My installation was successful. Thank you very much for your support.

Hello dear DavidGom1;

After completing the installation, I did the calibration process directly. Therefore, I could not pay attention to the calibration before. However, in the current situation, it starts from 0 in all 3 axes. It decreases as -1,-2,-3..... or increases as 1,2,3,4.... I do not see the 360 ​​value at all. It might be worth trying the updates in the link provided by dlktdr. Of course, if your card type fits.

rs3hc, 2024-09-26:
Yes, it works 👌

macgyverek, 2024-10-05:
Windows GUI for 2.4 not working. Could you update for download ?

dlktdr, 2024-10-05:
Looks like i need to update the action. Its deprecated.

Can use this in the meantime.
https://firmware.headtracker.net/actions/bin/1719763936.zip

dlktdr, 2024-10-22:
Actions are updated and running again

myguy98, 2024-11-22:
I purchased the Arduino Nano33 BLE REV2. I'm trying to get it to work with the latest Windows Gui version tally-Ho headTracker software. I've loaded and tried both the latest firmware version (Arduino Nano 33 Sense Revision 2 -v2.21), and (arduino_nano_33_ble_rev2-2_40pre-b89c1b2.bin) on my nano to no avail. Can someone direct me as to what I need to do to get my headtracker working? Thank You.

dlktdr, 2025-05-06:
Please look here for v2.4 firmwares.

https://github.com/headtracker/HeadTracker/actions?query=branch%3Av2.4x

closed 2025-05-06 by dlktdr