headtracker/HeadTracker #181: DTQSYS running version 2.21 has constant pan drift

igdianov, 2024-09-26:
I am observing constant pan drift on my DTQSYS head tracker running with 2.21 firmware while sitting still on my desk. The drift velocity is proportional to the gain value, i.e. at the highest gain setting, it drifts fully to the max within 2 min. With 3 us/deg gain it drifts slowly to max within 2 hours.

I tried to re-calibrate the accelerometer with the UI wizard while the magnetometer is disabled, but it didn't help. The roll and tilt remain constant, while pan axis drifts constantly while connected to UI. The same thing happens during flight, but the pan simply quickly shifts to the side after 2 minutes.

I tried to flash 2.40pre firmware, but the device stops working and UI is not able to connect to it.

LeonRemidy, 2024-10-07:
I have the same issue and hoping someone can help

Redvsbluecom, 2025-03-14:
Did anyone ever get this sorted?  I'm having the same problem with Ble33 v2.  Running 2.4 pre.  The pan slowly goes to the max unless I recenter

ProfDrYoMan, 2025-03-14:
Latest 2.4 pre with mag enabled is rock solid for me.
Ensure that you do not move the part when powering on.

Andry9550, 2025-04-17:
DTQSYS. Constant drift along the PAN(z) axis. The latest 2.4pre firmware did not help. It's getting worse with the magnetometer. Help!