headtracker/HeadTracker #186: Center button not working BLE Rev 2 board

Spudufly, 2024-11-07:
Hi - I Have everything working great on my HT except the centre button as wired in option 2.  The connections are correct and the continuity is correct - that is normally open / closed when pressed.  I can't see any settings to enable this - I'm using your new Rev 2 GUI and  arduino_nano_33_ble-2_40pre-7ae7247.bin firmware.  Anything you can suggest please?

SungwaYu, 2024-11-07:
I had the firmware version from #178 . For some reason the botton was set to D3 and cannot be changed. It worked for me when I wired it to D3 (the pin above the D2 that showed in the option 2 diagram).

dlktdr, 2024-11-07:
You can run the most recent version available under actions tab above, choose the newest ones compiled in the master branch. There will be a win gui and a firmware one. Rather than the older one from 178.

Yes the development version the pins are fixed and more people were using d3. A micro switch fits nicely across d3 and gnd in the popular cases. 

As soon as I get around to posting a release it I'll have to update the drawing too. I do intend on posting a firmware with D2 as well since there are also lots that followed that drawing. For time being will either have to rewire it or custom compile a firmware with the D2 as center using gitpod.

Spudufly, 2024-11-08:
Re wired to d3 and now works perfectly.  Thanks very much - brilliant!!

closed 2024-11-08 by Spudufly