headtracker/HeadTracker #187: ArduinoNano 33BLE Rev2

AP5RC, 2024-12-11:
Hello

I have just installed V2.2 on a NANO33 BLE Rev2 out of the box but I have no axis movement  or any  value info in the boxes.
I can see the green light on "Gyrometer Calibrated", the yellow diamonds in the pan/tilt/roll range but no activity when I move the card. I tried reinstalling the firmware few times but it didn't change anything.
My on-screen display looks same as one of the member on here who had same problem as I am having.  (Deniz-Derin 1St Sept) 
 Following online video/ Github book instructions as I am not good with computers. 
 Am I missing out anything? 
Has anyone got any ideas? 
Please assist.

AP5RC, 2024-12-14:
Solve the problem. Got the board working. If anyone having problem than log in Github here first. Click on  Head Tracker on top heading. Click on sub heading Actions. Under All workflows, click on Firmware Build, (amended last month)
download & use. For Windows download the from same list Windows GUI and use to program the board.(2months)

Jonathan-Ott, 2025-05-06:
> Solve the problem. Got the board working. If anyone having problem than log in Github here first. Click on Head Tracker on top heading. Click on sub heading Actions. Under All workflows, click on [...]

So use Firmware build #252, or can we use the newest (#271 at the time of this post)?

repusman, 2025-05-07:
> > Solve the problem. Got the board working. If anyone having problem than log in Github here first. Click on Head Tracker on top heading. Click on sub heading Actions. Under All workflows, click on Firmware Build, (amended last month) download & use. For Windows download the from same list Windows GUI and use to program the board.(2months)
> 
> So use Firmware build #252, or can we use the newest (#271 at the time of this post)?
> 
> Also, will this work on newer Rev2 boards (like 00072)?

I got it to work on 00071.  Both configurator and FW have to be from the same nightly update.