headtracker/HeadTracker #192: the IMU output-local board, the pan value continues to increase by 1 degree per second.

yangshourong, 2024-12-18:
[image]

The equipment is stationary.  
The values in the DIAGNOSTIC DISPLAY are stable, with no issues.  
However, in the IMU output-local board, the pan value continues to increase by 1 degree per second. 
[image]

How should I resolve this issue?

yangshourong, 2024-12-18:
[image]

yangshourong, 2025-01-03:
> Without magnetometer (absolute compass heading info), the pan will drift over time, as integrating acceleration has computational errors that accumulate over time.

Is there a way to solve this problem through software algorithms? Is there a plan to update the firmware to solve this problem?

rotorman, 2025-01-03:
There is no software solution for this - another ground truth-based reference system is needed, e.g. a compass/magnetometer, optical system or the like.
Think of this as like walking blindly in a large open area without further input - you will never be able to walk a straight line blindly without some correction from further input modality based on your "IMU" in your ears only.