headtracker/HeadTracker #195: Headtracker 2.4pre Xiao nrf52840 Non-sense(basic) can not receive data from ysoldak headtracker Xiao nrf52840 Sense module

Corey-Harding, 2025-01-11:
I get the following message over and over when using a xiao nrf52840 non-sense connected to a ysoldak headtracker Xiao nrf52840 Sense module, the drift is so much less with that other fork no matter what I try.  When both modules are on your fork it works as expected but I have drift.

```[21117504] <inf> btpararmt: Scanning successfully started
[21118231] <inf> btpararmt: Connecting to device...
[21123381] <inf> btpararmt: Connected: MAC (random)
[21123381] <inf> btpararmt: PHY Connection Rx:1 Tx:1
[21123382] <inf> btpararmt: BT Connection Params Int:10 Lat:0 Timeout:100
[21123388] <inf> btpararmt: Requesting coded PHY - Success
[21123850] <inf> btpararmt: Found FRSky Service 0xFFF0
[22106903] <err> bt_att: ATT Timeout for device MAC (random)
```

I am able to connect fine to he ysoldak headtracker xiao nrf52840 sense module using an esp32c3 and this fork of your fork of BTWifiModule (https://github.com/Corey-Harding/HeadtrackerRX)

I would like to use sbus into the radio though and also run another nrf52840 instead of an esp32c3 and don't want to do the work to this fork (https://github.com/Egor812/HeadtrackerRX) if I am missing something easy.

Corey-Harding, 2025-01-14:
Im not home but looking in my phone at your past github actions I see the build had failed.

'''/src/firmware/src/src/btjoystick.cpp:214:25: error: 'BT_LE_ADV_CONN_ONE_TIME' was not declared in this scope; did you mean 'BT_LE_ADV_OPT_ONE_TIME'?
  214 |   err = bt_le_adv_start(BT_LE_ADV_CONN_ONE_TIME, hidad, ARRAY_SIZE(hidad), NULL, 0);
      |                         ^~~~~~~~~~~~~~~~~~~~~~~
      |                         BT_LE_ADV_OPT_ONE_TIME'''

dlktdr, 2025-01-14:
Yeah it's a new feature that was in Zephyr that's not in the Docker image I've been using for actions.

Either you have to compile it using gitpod or local and update Zephyr before you compile or use the pre-compiled one that I posted on discord 

Keep meaning to update the image but I haven't

Corey-Harding, 2025-01-14:
Thanks. Will check it out later tonight.

Corey-Harding, 2025-01-14:
I tested the joystick branch on a nano ble rev 2 and it did not connect as expected to the ysoldak tracker.

However this branch, (https://github.com/dlktdr/HeadTracker/tree/xiao_52840_receiver), tested on the basic non-sense xiao ble does perform as expected and connects to the ysoldak tracker. In a little while I am going to solder the connections to the board for my sbus wires and test it out on a Radiomaster MT12.  I imagine it will function as intended at this point.

This solution is good enough for me, but moving forward somewhere there is a commit that broke compatibility with the ysoldak project if that happens to be something that is important for you to look into.

dlktdr, 2025-05-06:
I know this is old, I will re-open if needed. Please test using this firmware

https://github.com/headtracker/HeadTracker/actions?query=branch%3Av2.4x

closed 2025-05-06 by dlktdr