headtracker/HeadTracker #200: Developer documentation doesn't specify Zephyr version, which can cause build issues on some targets

MiguelFAlvarez, 2025-02-24:
When following online documentation to set up development environment, I was running into issues compiling the xiao_ble_nrf52840_sense.
I ended up finding out that my Zephyr version was set to 4.0.0 rc2. 
I noticed github actions build file specifies Zephyr 3.7.1, so I force installed that version. This seems to have addressed my build issues.
I also noticed you're using Zephyr SDK 0.17.0 instead of 0.16.5-1, so I switched to that. Not sure that mattered, but maybe is also worth updating on the gitbook page documentation.

dlktdr, 2025-02-24:
Yes. For sure it should be updated. I fixed the version in the workflows and the dockerfiles so hopefully development going forward wont have as many version mismatches. Although need to check the mac build still.

If your up for it can make prs on the docs too. If not will try to get to it later this week.

https://headtracker.gitbook.io/head-tracker-v2.2/radio-setup/how-to-edit-documentation

MiguelFAlvarez, 2025-02-24:
I'd rather leave that one to you. West and zephyr are all new to me, and I'm sure it took me more steps than necessary to get that setup correctly 😂. I already got the environment working on my end so no rush here, just figured would be good to know for anybody else in the future trying to build.

dlktdr, 2025-02-24:
Sounds good

dlktdr, 2025-03-30:
It has been added. 

https://app.gitbook.com/o/-Md3r5nfSt9hl-HcRAzT/s/ifYnTDP7IFJCZallDj96/support/development/firmware-using-west#get-the-zephyr-source-and-update

closed 2025-03-30 by dlktdr