headtracker/HeadTracker #205: Bluetooth Trainer signal

Brownsy67, 2025-04-10:
Bluetooth trainer signal lost when PAN past 1700us. Not always. Usually happens when moving slower, will reinitialize when moved other direction. 

NANO 33 BLE rev2.

C3-mini Bluetooth module, using the firmware from your BTWIFI project.

dlktdr, 2025-04-11:
What is the git tag of the version your running?

[image]

Can you send the settings you have too. Connect to the board then choose file->Export settings. Then copy that file here

[image]

Will see if I can reproduce it here

Brownsy67, 2025-04-11:
[image]

[General]
Git=1bfc991
Hard=NANO33BLE
Vers=2.40pre
accxoff=0.00152
accyoff=-0.00747
acczoff=0.00505
alertch=-1
an0ch=-1
an0gain=310
an0off=0
an1ch=-1
an1gain=310
an1off=0
an2ch=-1
an2gain=310
an2off=0
an3ch=-1
an3gain=310
an3off=0
aux0ch=-1
aux0func=0
aux1ch=-1
aux1func=0
aux2ch=-1
aux2func=0
btmode=1
btpairedaddress=
butlngps=false
ch5arm=false
crsftxinv=false
crsftxrate=140
dismag=true
gyrxoff=0.208331
gyryoff=-0.014023
gyrzoff=-0.242167
magxoff=0
magyoff=0
magzoff=0
pan_cnt=1500
pan_gain=5
pan_max=1950
pan_min=1050
panch=1
ppmchcnt=8
ppmframe=22500
ppmininvert=false
ppmoutinvert=false
ppmsync=350
pwm0=-1
pwm1=-1
pwm2=-1
pwm3=-1
rll_cnt=1500
rll_gain=5
rll_max=1950
rll_min=1050
rllch=2
rotx=0
roty=0
rotz=0
rstondbltapmax=400
rstondbltapmin=100
rstondbltapthres=80
rstondblttap=false
rstontlt=false
rstonwave=false
rstppm=-1
sbininv=true
sboutinv=true
sbustxrate=80
servoreverse=2
so00=0
so01=0
so02=0
so10=0
so11=0
so12=0
so20=0
so21=0
so22=0
tlt_cnt=1500
tlt_gain=5
tlt_max=1950
tlt_min=1050
tltch=-1
uartmode=0

Brownsy67, 2025-04-11:
[nano33rev2headtracker.zip](https://github.com/user-attachments/files/19699530/nano33rev2headtracker.zip)